timer-chefs/kitchen-timer #48: SPI communication between display and MCU

nperozzi, 2024-07-22:
I want to make the communication between the display and the MCU SPI since MAX7219 supports it. I believe we talked about it and you agreed. Unfortunately, I believe that it will require some changes in the max72xx library and display library that you already created.
I have started a branch to work on this: [SPI_Max72xx library](https://github.com/emanuelen5/kitchen-timer.git) 
This is how I am thinking about it:
* __SPI library__: just to initialize the SPI and a function to transmit a byte.
* __max72xx library__: to initialize the max7219 and functions to write to the registers

With the above, I believe this issues can be closed but, it got me thinking on the following two libraries. I can open a separate issue for this part, if that is ok.
* __display library__: to initialize and write to all max7219. This library should include a function that parses the patterns (numbers and other patterns) to the display.
* __patterns library__: to store the different patterns that can be displayed.

emanuelen5, 2024-07-23:
Just to clarify, it is already using SPI to the display, just not the hardware block in the MCU. By switching to this, we can free up lots of CPU cycles, which is great!

Related PR note: https://github.com/emanuelen5/kitchen-timer/pull/29#issue-2317580666
> ### Note
> We might want to switch this to the hardware SPI block in the future since it takes quite some time to send some of the SPI commands. Sending a single bit takes about 8 clock cycles. Each command is 16 bits. And we have four devices on the SPI bus. This gives a total of $8 \times 16 \times 4 \approx 0.5 \text{ ms}$. Say we want to update the whole displays, then we have to send 8 commands: $8 \times 16 \times 4 \times 8 \approx 4.1 \text{ ms}$!
> 
> We can probably use the SPI hardware block if we want to offload the CPU from the data shifting. But, then we need to connect DIN of the MAX7219 chip to pin 17, which is shared with the programmer. According to some sources this shouldn't conflict with the programmer[^1] since the MOSI pin becomes an input when the RESET signal is asserted.[^3]

> [^1]: https://forum.arduino.cc/t/spi-miso-mosi-used-in-sketch-possible-to-re-program-via-external-programmer/492461/8
> [^2]: https://onlinedocs.microchip.com/pr/GUID-EEA42155-A873-46B4-9500-D88617041B0B-en-US-2/index.html?GUID-BB147C25-35E4-42CA-BE16-926534E4EC12
> [^3]: https://onlinedocs.microchip.com/pr/GUID-EEA42155-A873-46B4-9500-D88617041B0B-en-US-2/index.html?GUID-BB147C25-35E4-42CA-BE16-926534E4EC12

nperozzi, 2024-07-23:
I think I am stepping on your toes here 😟 . Last night I basically re-wrote your max72xx library when maybe it would suffice with some changes (see [SPI_Max72xx library](https://github.com/emanuelen5/kitchen-timer.git)).
How would you like me to proceed here? Maybe I should let you add the hardware SPI block yourself?

nperozzi, 2024-08-17:
When possible, we should take a look at your max72xx and display libraries. I don´t understand what is happening there.

I am also not sure what you mean by the "SPI block". I am guessing that you mean that the max7219 is receiving SPI but the message is generated by the MCU instead of using the microcontroller´s SPI functionality.

PR #50 (2024-10-08): Use the SPI hardware block instead of bit-banging
Closes #48
closed 2024-10-08 by emanuelen5