timer-chefs/kitchen-timer #65: What should be the track width and the via size?

nperozzi, 2025-05-09:
# Track width:
I used the KiCad Calculator Tool. For an avg. current draw of 85.5mA, as calculated in the schematic (sheet Battery Power Requirements), the signal tracks' width should be 9.69mils (~10mils = 0.254mm).

[image]

For the power nets, I used the peak current draw of 1278.5mA. In this case, the power track width should be at least 16.9mils (~17mil = 0,43mm).

[image]
 
#Via size
We are using a via size of 0.7mm in diameter and 0.3mm hole.
We used the KiCad Calculator Tool. The resistance, voltage drop, and power loss seem negligible for this via size. 

[image]

nperozzi, 2025-05-09:
- [ ]  We need to adjust the track size for power.

emanuelen5, 2025-05-10:
Great! So neither of the tracks are under-dimensioned as it is. Having a larger power track is also better, even if we would make do with a smaller one.

> We are using a via size of 0.7mm in diameter and 0.3mm hole

We're actually using 0.8 mm since the DRC didn't like the annealing width when it was set to 0.7:

[image]

Sounds like we don't have to make any adjustments?

emanuelen5, 2025-05-10:
It looks like you used 855mA in your calculation, not 85.5?

[image]

nperozzi, 2025-05-11:
> > I used the KiCad Calculator Tool. For an avg. current draw of 85.5mA, as calculated in the schematic (sheet Battery Power Requirements), the signal tracks' width should be 9.69mils (~10mils = 0.254mm).
> 
> It looks like you used 855mA in your calculation, not 85.5?
> 
> [image]

You are right. It was a mistake. Here is the calculation with 85.5mA. It is way smaller in width, so we are good anyway. :) 

[image]

closed 2025-07-31 by nperozzi